Zanduino/DS3231M #11: The SETDATE typed input fails due to the inputBuffer being shared (on Arduino Uno)

simmunity, 2019-10-16:
On the slow Arduino Uno, the SETDATE function always fails as you've shared the input buffer with the sprintf displaying the date/time. On fast CPU's this might work, but not on an Uno.

Adding:
char outputBuffer[SPRINTF_BUFFER_SIZE]; // right below the inputBuffer definition

Changing inputBuffer to outputBuffer:
    sprintf(outputBuffer,"%04d-%02d-%02d %02d:%02d:%02d", now.year(),          // Use sprintf() to pretty print    //
            now.month(), now.day(), now.hour(), now.minute(), now.second());  // date/time with leading zeros     //
    Serial.println(outputBuffer);  

Fixes the problem. The latest example on the Arduino library has this error.
Otherwise this is a nice library. I'd send a pull request but the changes are soo minimal.
Thanks, Shannon

SV-Zanshin, 2019-10-16:
Hello  Shannon,
I am assuming you mean the "sprintf" at line 175 in the "Set.ino" example program.
Since this sketch doesn't use interrupts, any execution of code will be linear and the CPU speed won't affect the order in which things happens, just their timing. The "Serial.println()" command after the "sprintf()" won't return until the whole buffer has been sent to the serial port so there's no chance of garbling the "inputBuffer". I tested this on several Arduinos (UNO, Micro (32U4 processor), Mega, and other processors on a breadboard) with no problems so I'm not quite sure what problem you are seeing. 
I do think  I should refactor the program and rename "inputBuffer" to "textBuffer" so that it is clear that this character buffer is being used for both types of functions.

SV-Zanshin, 2019-10-16:
Hello Shannon,

I looked at the code again and realized that I was wrong, the "readCommand" doesn't always read the whole input string in one iteration and when that happens the output of the date can be mangled. I will change the code around to fix that.

closed 2019-10-16 by SV-Zanshin