Zanduino/DS3231M #12: the bug of HOUR writing&reading

neocao123, 2019-11-11:
There's some bug when hour count's more than 20.After reading the reference of DS3231M,Knowing that HOUR isn't typical BCD,I suggest do so:
    the low-5-bits are just like BCD;
    read()&B01000000: 0 is 24h system,1 is12h-sys;
    in 24h-sys,B00100000 means 20+,and attention that B00110000 is not allowed;
    in 12h-sys,read()&B00100000:0 is AM,1 PM.

for example:
   23:B00100011 or B01110001,not B00110011;
   13:B00010011 or B01100001;
   11:B00010001 or B01010001;
   1:B00000001 or B01000001.

SV-Zanshin, 2019-11-12:
I understand the problem as you've described. I will modify the BCD conversions for Hours in the code and test the results. It would seem that I did all of the development and testing for this part of the code during the wrong hours, otherwise I would have noticed that error!

SV-Zanshin, 2019-11-14:
Sorry it is taking so long - I can't seem to find my DS3231M in my boxes of ICs.

SV-Zanshin, 2020-05-17:
I found my DS3231M boards and have tried to reproduce the problem you opened, but without success. Using the example program "set.ino" to set the time, I can successfully set the date and time to any legal value.

Could you explain how I can reproduce the issue?

SV-Zanshin, 2020-05-17:
I see - the datetime() function has no range checking, so entering a larger hour than 23 allows the device to count up values that are incorrect. I've added range checking for the time (clamping it between 0 and 23) to fix this problem.

closed 2020-05-17 by SV-Zanshin