Zanduino/DS3231M #17: .adjust() not setting the correct time

jegjessing, 2021-03-18:
`Hi, I'm using code from the example adjusting the time:
 Serial.print(F("\nStarting Set program\n"));
  Serial.print(F("- Compiled with c++ version "));
  Serial.print(F(__VERSION__));
  Serial.print(F("\n- On "));
  Serial.print(F(__DATE__));
  Serial.print(F(" at "));
  Serial.print(F(__TIME__));
  Serial.print(F("\n"));
  DS3231M.adjust();

now = DS3231M.now();
    Serial.print(F("Time: "));
    Serial.print(now.hour());
    Serial.print(F(":"));
    Serial.print(now.minute());
    Serial.print(F(":"));
    Serial.println(now.second());`
But it does not seem as if it is working, because I get something like:
_08:48:48.584 -> - Compiled with c++ version 7.3.0
08:48:48.584 -> - On Mar 18 2021 at 08:47:52
08:48:48.584 -> Time: 1:16:3_
The compiled with time is correct, but the actual time is way off. Almost seem random!?

jegjessing, 2021-03-18:
Tried to format the code lines, but that would not work for me :-(

SV-Zanshin, 2021-03-18:
Hello!

I vaguely remember having seen that problem before, but can't recall the details.  Are you using the most recent library version and which Arduino are you using to reproduce the problem? If you use the adjust function to set an explicit time, e.g. "DS3231M.adjust(DateTime(2017,8,5,18,19,20));" does that work?

jegjessing, 2021-03-18:
Hi, happy to hear that it's been seen before :-)

I'm using the most resent library yes.
It is both the Uno and the Mega ones.

Using the example you gave above the time was set to:
`11:10:42.488 -> Time: 18:19:20`

So, I would say yes. Works when set manually

jegjessing, 2021-03-18:
Cold it be some locale stuff?

SV-Zanshin, 2021-03-18:
I seem to recall it had to do with the way that the compiler stores the date/time in the program and thought it might only have happened on the ESPlora processors.  I will search through my cupboard for a functioning DS3231M and test it; but that might take a couple of days... (that's the downside of "free" support :) - nothing happens fast)

jegjessing, 2021-03-18:
Actually I changed it to: DS3231M.adjust(DateTime(__DATE__, __TIME__)); and that works fine.. 

So I'm kinda ok now. But maybe the .adjust() should be modified to work with the above?

And no rush :-) U'm just happy someone reacted. Buying products is definitely not a guarantee for as good support as on open source projects.

SV-Zanshin, 2021-12-12:
Sorry for the long delay, I was overseas and didn't have a DS3231M to test with. I'm back now and haven't located a DS3231M to play with yet, but I think the problem lies on the compiler side. Are you compiling on a Windows or a Linux machine?

SV-Zanshin, 2021-12-12:
What is display if you use __VERSION__, __DATE__, and __TIME__ pre-processor macros? Those are the officially supported ones for the compile, in my installation "DATE", "TIME" and "VERSION" aren't declared as macros.

lagg070988, 2021-12-13:
I have reproduced the code and I get the same error in ubuntu 20.04 arduino 1.8.16 library 1.0.8 arduino nano, I have added some serial.print for (fun and in the spirit of learning) and discovered the following: 
I have seen that: `DS3231M.adjust ();` becomes `adjust (DateTime (F (__ DATE__), F (__ TIME__))); line 323 ` 
then `DateTime::DateTime(const __FlashStringHelper* date, const __FlashStringHelper* time); line 146` 
then in line 160, (calling a constructor within another constructor): ` DateTime(ybuff, tbuff)`  
and here is when the DateTime object is generated. So far everything is fine, the object retains the date and time as it should be, but when going back out of `DateTime::DateTime(const __FlashStringHelper* date, const __FlashStringHelper* time);` the object disappears and contains no data.
and `void DS3231M_Class::adjust(const DateTime& dt); line 325` receive nothing (or random values )
doing ` DateTime agua = DateTime (F(__DATE__), F(__TIME__));   Serial.println(agua.year());` get nothing consistent.
I'm too new to c ++ and I still don't understand what happens with the disappearing object, it seems to me that when calling a constructor inside another, it creates an object that is only available inside it, and cannot be used outside. but I'm not sure, I'm just learning. hope this can help.

lagg070988, 2021-12-15:
solve by adding this on line 160 
`  DateTime a = DateTime( ybuff, tbuff);     // Use the string version to instantiate`
`  yOff = a.yOff, m = a.m, d = a.d, hh = a.hh, mm = a.mm, ss = a.ss;`
But something curious happens: the saved value of `__date__` and `__time__` from the library is updated every time I make a change in the library, or every time I restart the Arduino IDE.
If in a session I upload the sketch 3 times, they will all have the same value for `__date__` and `__time__`
it must be an arduino IDE matter

SV-Zanshin, 2021-12-18:
I can't reproduce this with the Windows compiler and IDE.  The __date__ and __time__ are compiled into the library when it is built. It could be that your IDE settings rebuild the library automatically when the IDE is opened, which would explain the behavior.

I think this problem is really down to compiler settings and differences on various platforms.

lagg070988, 2021-12-26:
I have reproduced the code and I get the same error in:
- ubuntu 20.04 arduino 1.8.16 library 1.0.8 arduino nano (my pc)
- Windows 10 arduino 1.8.19 library 1.0.8 arduino nano (my son's pc)
- Arduino online editor (https://create.arduino.cc/editor)

works as it should when adding this on line 160:
```
DateTime a = DateTime( ybuff, tbuff); // Use the string version to instantiate
yOff = a.yOff, m = a.m, d = a.d, hh = a.hh, mm = a.mm, ss = a.ss;

```

> I can't reproduce this with the Windows compiler and IDE. The **date** and **time** are compiled into the library when it is built. It could be that your IDE settings rebuild the library automatically when the IDE is opened, which would explain the behavior.

this is curious, I have tried in 3 different ways and I get the same result

PR #21 (2021-12-27): Update DS3231M.cpp
Fixes #17

- [ ] Bug fix (non-breaking change which fixes an issue)

[[image] [image]](https://zanduino.github.io)
closed 2021-12-27 by SV-Zanshin