Zanduino/DS3231M #18: DS3231 vs DS3231M

lagg070988, 2021-11-17:
Hello, I realize that I have in my possession a DS3231, and it has some features that the DS3231M does not have.
One of them is the ability of the DS3231 to produce a frequency of 1hz, 1khz, 4khz or 8khz. while the DS3231M only produces 1hz.
[here is a small explanation](https://blog.heypete.com/2017/09/05/major-differences-between-the-ds3231-and-ds3231m-rtc-chips/),  although I am guided by the data sheets.
I am fairly new to github, I "fork" your library and made some add-ons (some lines of code) to support this feature. and it has worked fine on arduino uno, nano and some boards that I assemble as data logger.
since this library appears in arduino ide, I wonder: would it be wise to include these changes and expand the support for DS3231?

SV-Zanshin, 2021-12-12:
I have made a some changes to the code which make this pull request a bit of a chore to integrate, so I am going to add your code to my branch and integrate that..

SV-Zanshin, 2021-12-12:
I've added the changes, having changed the code just a little bit to stay backwards-compatible. I'll draft a new 1.0.7 release that will include this additional functionality as well. Thanks!

closed 2021-12-12 by SV-Zanshin

lagg070988, 2021-12-12:
thanks for you!
 I believe that  writeByte(DS3231M_CONTROL, readByte(DS3231M_CONTROL) & ~**B00111000**); line 528
should be writeByte(DS3231M_CONTROL, readByte(DS3231M_CONTROL) & ~**B00011100**);

in writeByte(DS3231M_CONTROL, (readByte(DS3231M_CONTROL) & ~**B00111000**) | (rate | B00000011) << 3); line 540
should be  writeByte(DS3231M_CONTROL, (readByte(DS3231M_CONTROL) & ~**B00011100**) | (rate | B00000011) << 3);
in the next few days I will prove this.

SV-Zanshin, 2021-12-12:
You are correct, I don't know how that slipped by me!

lagg070988, 2021-12-12:
I forgot to highlight the change between the "or" and "and" in the previous comment, line 540, otherwise the output would always be 8khz
sorry

lagg070988, 2021-12-13:
I have tested today on arduino nano and it works as expected pinSquareWave () and pinSquareWave (const uint8_t rate) .
thanks!

SV-Zanshin, 2021-12-14:
That's great news - I'm glad we could make the library work better.