Zanduino/DS3231M #24: How to stop an defined alarm

capitainekurck, 2023-05-09:
Hi

I got a problem, I can't find how to stop a defined alarm. 
i work on a project where i need to set an alarm for a period for example every minutes between 01h00 and 03h00
I set an alarm like this:
DS3231M.setAlarm(everyMinute,DateTime(2023,6,9,01,00,0));

but after i can't stop it it fire every minutes. i tried to stop it with the parameter "alarmState":
DS3231M.setAlarm(everyMinute,DateTime(2023,6,9,01,00,0),false);
but it do nothing the alarm still fire every minutes. I can't find how to clear it (to stop the alarm fire)
the clearAlarm() only clear the fire flag for the next fire.

on the RCTlib from adafruit there is an API wo call disableAlarm() who seem to do the job, but i can't find such api on your library.
i use your library because it seem to be lighter than adafruit one.

regards

SV-Zanshin, 2023-05-14:
I see what you mean; you've done the correct thing by setting the alarm state to FALSE in the SetAlarm() method.  

Looking at my code for the method isAlarm(), I don't check whether or not that bit is set. And since the 2 bits for the alarms keep their old values, this results in the problem you see.

I'm going to make some changes to the isAlarm() method in the library.  I don't have a DS3231M handy to test my logic, can you check to see if it is now working?

SV-Zanshin, 2023-05-14:
I've made the changes to the isAlarm() method. Can you add that to your code and see if that correctly resolves the alarm problem?

capitainekurck, 2023-05-15:
Hi,
I try the new version, there is a problem, now the alarm is fire on every test : if (isAlarm() ...
like the isAlarm() is always true even with the clearAlarm()

SV-Zanshin, 2023-05-15:
That's not good :(
I will look at the code again; it sounds like I've mixed up some && and || conditions...

SV-Zanshin, 2023-05-15:
OK, now I think I've done it correctly, I also simplified the logic to

uint8_t controlByte = readByte(DS3231M_CONTROL) & 0x02;   // Mask out all but last 2 bits for ALM1&2
uint8_t statusByte = readByte(DS3231M_STATUS) & 0x02;        // Mask out all but last 2 bits for ALM1&2

The last two bits of both variables contain corresponding bits for "active" and "alarmed", so "and"ing them together will give either 0 for no alarms, or 0,1 and 2 (11b, 10b, 01b) for the alarm that is set.

SV-Zanshin, 2023-05-15:
Sorry, I didn't want to close this until you confirm that it is working.

capitainekurck, 2023-05-15:
hi,

I will do more test but for the first test it seem to work :+1:

SV-Zanshin, 2023-05-17:
That's good news. I think I'll draft a new version that incorporates this fix.

closed 2023-05-17 by SV-Zanshin