Zanduino/DS3231M #25: {"Add Short one-line Bug title"=>"STM32 cant upload again"}

DprasK, 2023-07-04:
## Expected Behavior

_Detail what the expected program behavior should be, or what is expected to happen._

## Actual Behavior

_Detail what actually happens._

## Steps to Reproduce the Problem

_Explain what needs to be done in order to reproduce the problem._
  1. ST-Link can't detect again after uploading this library
  2.
  3.

## Specifications

  - Library Version: 1.0.11
  - IDE Version: Arduino 2.1.0
  - Platform: STM32F4VE
  - Subsystem:
  - _any other details needed to reproduce the problem_

SV-Zanshin, 2023-07-04:
I don't have a STM32F4VE with which to test this, so I cannot look into this issue. 

The library makes use of the "wire" library but otherwise has no code specific to the Atmel processors. So there is no general reason why this shouldn't work. 

I do see several entries regarding communications problems with ST-Link, so I would tend to believe that the source of the issue is there. Do you have another project that uses the "wire" library that does work on your STM32F4VE?

DprasK, 2023-07-05:
LiquidCrystal_I2C, RTClib (work on maple board, but on official board cant work), ds323x